smdn/Smdn.Devices.Mcp2221A #1: [Linux][HIDSharp] System hangs while I2C read

smdn, 2021-05-13:
HIDSharp problem?

# repro code
```cs
using System;

using Microsoft.Extensions.DependencyInjection;
using Microsoft.Extensions.Logging;
using Microsoft.Extensions.Logging.Console;

using Smdn.Devices.MCP2221;

var services = new ServiceCollection();

services.AddLogging(
  builder => builder
    .AddSimpleConsole(static options => options.SingleLine = true)
    .AddFilter(static level => LogLevel.Trace <= level)
);

await using var device = await MCP2221.OpenAsync(services.BuildServiceProvider());

var address = new I2CAddress(0x20); // MCP23017

await device.I2C.WriteByteAsync(address, 0x12); // GPIOA

var buffer = new byte[2];

await device.I2C.ReadAsync(address, buffer); // System hangs here

Console.WriteLine(BitConverter.ToString(buffer));
```

```dotnet build /p:UsbHidDriver=HIDSharp && stdbuf -o0 dotnet run --no-build &> test.log```

# log
test.log:
```
(no log recorded)
```

/var/log/syslog:
```
(no log recorded)
```

# env
```
.NET SDK (global.json を反映):
 Version:   5.0.202
 Commit:    db7cc87d51

ランタイム環境:
 OS Name:     ubuntu
 OS Version:  20.04
 OS Platform: Linux
 RID:         ubuntu.20.04-x64
 Base Path:   /usr/share/dotnet/sdk/5.0.202/

Host (useful for support):
  Version: 5.0.5
  Commit:  2f740adc14
```

wulinzhou, 2021-12-23:
i have test read and write function, it is some time ok for i2c
Buffer[0] = 0x00;
        Buffer[1] = 0x00;
        device.I2C.Write(wlcadd,Buffer,0,2);
        int num = device.I2C.Read(wlcadd,Buffer,0,16);
        Console.WriteLine("Read {0}",num);
        Console.WriteLine(BitConverter.ToString(Buffer));

smdn, 2021-12-23:
@wulinzhou Thank you for testing and reporting!