smdn/Smdn.Devices.Mcp2221A #3: [TODO] ADC functionality not implemented

smdn, 2021-05-13:
The interface is not definite.
https://github.com/smdn/Smdn.Devices.MCP2221/blob/1931ceb937a8e39c3dcc3f5cdf7ab1dbbb58f41a/src/Smdn.Devices.MCP2221/Smdn.Devices.MCP2221/MCP2221.GPs.ADC.cs#L10-L20

cartheur, 2022-12-15:
Is there a plan to implement this interface in the near future?

smdn, 2022-12-15:
I have no plans to implement this feature in the near future.
I'm currently working on another project and do not have time to spend on this project. Sorry for inconvinience.

I also read your email, and I'll reply here.

If you are currently writing code or planning to implement for ADC/DAC feature, and have any questions, please ask here in this issue. I will answer.

Or, if you have already implemented some codes for this feature, please send me a pull request. I would like to start working on the merge.
However, the pull request cannot be merged immediately. I will merge it after checking with the actual device.

Anyway, contributions are welcome. Please let me know what you are planning to do.