smdn/Smdn.Devices.Mcp2221A #8: ChipFactorySerialNumber returns 01234567

CZEMacLeod, 2021-11-12:
I am using an [Adafruit MCP2221A Breakout](https://www.adafruit.com/product/4471) on Windows 11 and am getting a connection working as expected and can access the [Adafruit SCD-40](https://www.adafruit.com/product/5187) I have connected without any problem, using `Iot.Device.Scd4x` from `Iot.Device.Bindings`, so I do not believe there is any issue there.

Also, when I retrieve the `HardwareRevision` and `FirmwareRevision` properties I get A.6 and 1.2 as expected.

However, the `ChipFactorySerialNumber` property always returns `01234567`.

I have checked with the Microchip supplied tool `MCP2221 Utility` and it shows the Serial Number for my unit.
```
Currently Selected Device Info
Device 1 out of 1 is selected.

Serial Number = 0003167395
Device Path = \\?\hid#vid_04d8&pid_00dd&mi_02#9&9255fb8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
```
Any help on this would be appreciated.

smdn, 2021-11-13:
Thank you for using my library and reporting an issue!

I have confirmed the same problem also occurs with my device.
It seems to return `01234567` as a value of `ChipFactorySerialNumber` always in some reason.

I will find out the reason of this problem and will fix some code.

smdn, 2021-11-14:
Sorry for the inconvenience.
I added new property `SerialNumberDescriptor`([diff](https://github.com/smdn/Smdn.Devices.MCP2221/commit/6d3536a833857cb4eeb8a21a6f3ba507cb8a89ec#diff-056271cba6ff8d350df7910f412c01490cd7e67256eba4c229beb927a96c6b53)) and released NuGet package [0.9.1](https://www.nuget.org/packages/Smdn.Devices.MCP2221/0.9.1).

In this version, `SerialNumberDescriptor` reports the serial number currently set to the MCP2221.
So If you just need to retrieve the *serial number*, please update and use it.

The reason why the `ChipFactorySerialNumber` reports `01234567` is not a problem with the library, but with the behavior of MCP2221.
MCP2221 always reports `01234567` and the reason for this behavior is unknown...

I will continue to investigate the problem about `ChipFactorySerialNumber`, so please wait for a future update if you need this value.

CZEMacLeod, 2021-11-14:
@smdn Thank you for the quick response. I had looked through the code and couldn't find the error, so I am glad it is external to the library.
The SerialNumberDescriptor field will be perfect for my use so I will update and try it out as soon as I can.

CZEMacLeod, 2021-11-14:
Using `SerialNumberDescriptor` is working for me - I don't know if you want to leave this open until you get an answer to the original issue with `ChipFactorySerialNumber ` returning `01234567`, but for my use case you can close it.

smdn, 2021-11-14:
I'm glad to hear that you were able to accomplish your goal.

I would like to close this issue for now.
If you need `ChipFactorySerialNumber`, please open the issue again!

closed 2021-11-14 by smdn